tritonrocketclub/Propulsion #1: Nozzle Throat-Exit ratio optimization

cmflannery, 2016-10-04:
Nozzle throat-exit dimensions need to be reviewed.

cmflannery, 2016-10-04:
starting work on nozzle optimization

Bslieu99019, 2016-10-11:
**Important Links**

1. [8-inch Motor Calcs](https://github.com/tritonrocketclub/Propulsion/tree/master/Launches/8-inch-flight/Design/Calculations/trunk)
2. [Nozzle wiki](https://github.com/tritonrocketclub/Propulsion/wiki/%5BDocumentation%5D-8-inch-Rocket)

Bslieu99019, 2016-10-28:
Aluminum 6061T-6 Carrier Calculations:

Hoop stress w/ FS = 2.0 => 12.8 ksi

Thermal Stress w/ 300 F & FS = 2 => 31.7 ksi (various with temperature)
Aluminum Melting point = 1080 - 1205 °F

Tensile Strength Aluminum = 40 kpsi
Minimum thickness w/ FS = 2: 0.133125 inches

Ultimate Bursting Pressure = 3000 psi
Yield Bursting Pressure = 2666.67 psi
Maximum Pressure Experienced = 710 psi

jacquelynfirst, 2016-11-01:
I worked on the aluminum carrier with Ryan and this is what we found so far. We got a thermal stress of about 55,000psi with a temp estimate of 430 degree F change. Also Hoop stress was 7,114psi with factor of safety of 5.6. 

Happy Halloween!

dtt005, 2016-11-03:
Best diameter for diverging section = 5.618in
Optimum Expansion Ratio = 6.177
Pressure experienced of aluminum skirt at exit plane = 16.546PSI (12.589% higher than atmospheric pressure)