tritonrocketclub/Propulsion #20: Liner fitting into airframe

cmflannery, 2016-10-14:
The tolerances on the liner and the aluminum casing need to be large enough to allow the liner to slide into the airframe. Hoop stress calculation and material calculations should be completed on the liner to ensure the pressure in the combustion chamber doesn't cause the liner to fracture.

Look into other possible solutions.
* o-rings

Bslieu99019, 2016-10-15:
10/14 - Hoop Stress Calculations:
-FS of 1.5: ~ 31k psi
-FS of 2.0: ~ 42k psi
-Material Interest: Garolic G10: 32-40k psi (max allowable)

cmflannery, 2016-10-29:
**Phenolic Liner**
Liner thickness; 1/8"
Liner Length: 7' 10" = 94"

**Casting Tubes:**
Thickness: 1/8"
Length: 15"

**Grains**
Length: 14"
OD: 6.90"
ID: 2.75"

rlibman, 2016-11-01:
**Phenolic Liner Hoop Stress Calculations**

Using http://www.engineeringtoolbox.com/stress-thin-walled-tube-d_948.html 

Hoop Stress = (745.9793426 psi * 7.062 in) / 2 * .25 in = 10536.2122348824 psi 

Per http://www.g10fr4.com/g10_fr4_technical_data_specifications.html 

19,000 psi max

Therefore a F.S. of ~1.8.

cmflannery, 2016-11-01:
The purpose fo the phenolic liner is to act as a thermal barrier between the combustion chamber and the aluminum casing. The phenolic liner doesn't need to take hoop stress. The heat transfer through the phenolic liner to the aluminum should be calculated and then the hoop stresses in the aluminum liner should be compared against its yield stress.